atomvm/AtomVM #1671: Problem testing `disterl` with connection from another Erlang node

weaversam8, 2025-05-19:
I tried to test the newly added distributed erlang support from the `main` branch. These are the steps I took:

1. Compiled AtomVM, and ran `./src/AtomVM examples/erlang/disterl.avm` to start node A
2. Started another node B with `erl -name node@127.0.0.1`
3. Ran `erlang:set_cookie('atomvm@127.0.0.1', 'AtomVM').` on node B,
4. Ran `{disterl, 'atomvm@127.0.0.1'} ! quit.` on node B.

The message sends successfully on node B, but AtomVM prints a crash message on node A and does not exit:

```
CRASH 
======
pid: <0.8.0>

Stacktrace:
[{socket_dist_controller,process_recv_buffer,3,[{file,"/Users/<username>/Git/AtomVM/libs/estdlib/src/socket_dist_controller.erl"},{line,192}]},{socket_dist_controller,recv_data_loop,1,[{file,"/Users/<username>/Git/AtomVM/libs/estdlib/src/socket_dist_controller.erl"},{line,175}]},{socket_dist_controller,handle_cast,2,[{file,"/Users/<username>/Git/AtomVM/libs/estdlib/src/socket_dist_controller.erl"},{line,151}]},{gen_server,loop,3,[{file,"/Users/<username>/Git/AtomVM/libs/estdlib/src/gen_server.erl"},{line,539}]}]

cp: #CP<module: 15, label: 55, offset: 37>

x[0]: error
x[1]: badarg
x[2]: {4,4,126,2,[{3,3322},{15,919},{15,1492},{15,1648}],error}

Stack 
-----

{<<112,131,104,4,97,6,88,119,20,115,119,101,97,118,101,114,54,53,51,64,49,50,55,46,48,46,48,46,49,0,0,0,87,0,0,0,0,104,43,111,231,119,0,119,7,100,105,115,116,101,114,108,131,119,4,113,117,105,116>>,<<"">>}
0
<<"">>
#CP<module: 15, label: 55, offset: 37>
[]
{state,{<<"">>,#Ref<0.0.12>},<<"">>,undefined,<<"">>,0,0}
#CP<module: 15, label: 39, offset: 69>
#CP<module: 3, label: 113, offset: 24>
[]
[]
[]
{state,{<<"">>,#Ref<0.0.12>},undefined,undefined,<<"">>,0,0}
[]
{state,undefined,socket_dist_controller,{state,{<<"">>,#Ref<0.0.12>},undefined,undefined,<<"">>,0,0}}
<0.7.0>
#CP<module: 3, label: 146, offset: 0>

Mailbox
-------

Monitors
--------
link to <0.9.0>
monitored by resource 0x132908048 ref=28

**End Of Crash Report**
```

pguyot, 2025-05-19:
Which version of Erlang/OTP are you using?

It seems that AtomVM doesn't understand the message that it is sent, hence the badarg error. I tried with OTP 27.

Do you have another message in addition to the crash log?

UncleGrumpy, 2025-05-19:
In the AtomVM example the cookie is set with:
```erlang
net_kernel:set_cookie(<<"AtomVM">>),
```
It is a binary string, on the OTP node you are using an atom.

I wonder if the AtomVM node is possibly not ignoring messages with a different cookie, but instead ignoring the contents of the message, resulting in trying to process an empty message?

UncleGrumpy, 2025-05-19:
I have generally set the cookie and the name when starting a node from the cli, so I believe that is interpreted as a binary string, you should be able use:

1. Compile AtomVM, and run ./src/AtomVM examples/erlang/disterl.avm to start node A
2. Start another node B with erl -name node@127.0.0.1 -cookie AtomVM
3. Run {disterl, 'atomvm@127.0.0.1'} ! quit. on node B.

UncleGrumpy, 2025-05-19:
But this does still reveal a bug in how messages sent with a different cookie are handled.

UncleGrumpy, 2025-05-19:
> Or in your original, substitute:
> 
> 3. Ran erlang:set_cookie('atomvm@127.0.0.1', <<“AtomVM”>>).
> 
> You could also rewrite the AtomVM node example to use an atom for the cookie, or in a real application, if that is your preference.

I just spotted another problem with your cli node. You gave it the same name, on the same host as the example node is using. This should be:

```erlang
erlang:set_cookie('node@127.0.0.1', <<“AtomVM”>>).
```

weaversam8, 2025-05-19:
Just to add some context, I was following the guidance that the test output when run, which looks like this:
```
% ./src/AtomVM examples/erlang/disterl.avm
Distribution was started
Node is 'atomvm@127.0.0.1'
Cookie is AtomVM
This AtomVM node is waiting for 'quit' message, and this process is registered as 'disterl'
On an OTP node with long names distribution, run:
erlang:set_cookie('atomvm@127.0.0.1', 'AtomVM').
{disterl, 'atomvm@127.0.0.1'} ! quit.
```

It doesn't look like it's an issue with an invalid cookie, because if I try to send a message without setting the cookie, I do get an error message and not a crash on the AtomVM side:
```
dist_util: shutting down connection dist_util:529, data={'node@127.0.0.1',invalid_challenge}, reason=shutdown
```

@pguyot - I was using Erlang `25.3.2.12`. Testing with Erlang `27.3` produces the same error. I do not have another message in addition to the crash log.

@UncleGrumpy - I think some of what you mention is incorrect.

1. [The erlang docs for `set_cookie/2`](https://www.erlang.org/doc/apps/erts/erlang#set_cookie/2) suggest that a cookie must be an atom, and trying to set a cookie to a binary creates an error message. 
2. [The docs for Distributed Erlang](https://www.erlang.org/doc/system/distributed.html) suggest that `set_cookie/2` can be used to set the cookie used to connect to another node (in this case Node A) even if its not the cookie your node uses.
3. `-cookie` should be `-setcookie` per the [`erl` command docs](https://www.erlang.org/doc/apps/erts/erl_cmd). Connecting using `erl -name node@127.0.0.1 -setcookie AtomVM` produces the same error.

UncleGrumpy, 2025-05-19:
You are absolutely correct. I just forgot that was one of the minor differences in AtomVM is that we use a binary string to set the cookie, but it is indeed an atom in OTP. But with the extra info I realized that you are not loading the libraries with the example, if you use the folowing fro inside the build directory it should work for you:
```bash
$ src/AtomVM libs/atomvmlib.avm examples/erlang/disterl.avm
```
If the AtomVM application is "installed" and you use the launcher script it will load the libraries for you.

pguyot, 2025-05-19:
Thank you for the report. Bug is confirmed and fixed by #1639 . There is a binary corruption happening in decoding the packet which is fixed by the externalterm fixes in #1639.

weaversam8, 2025-05-21:
Thanks for letting me know! I pulled down your branch and confirmed the fix worked. I did see these dangling resource log messages before AtomVM exited, but I suspect that's expected based on the current state of the project:

```
Warning, dangling resource of type socket_fd, ref_count = 6, data = 0x147a04e58
Warning, dangling resource of type socket_fd, ref_count = 3, data = 0x147b040a8
Warning, dangling resource of type socket_fd, ref_count = 4, data = 0x147d040a8
Warning, dangling resource of type dist_connection, ref_count = 2, data = 0x147811098
```

bettio, 2025-05-24:
#1639 has been merged. Closing this issue.

closed 2025-05-24 by bettio