atomvm/AtomVM #1677: documentation typo? `mix.esp32.flash` --> `atomvm.esp32.flash`

a-maze-d, 2025-05-25:
`mix.esp32.flash` is mentioned in the following lines:
https://github.com/atomvm/AtomVM/blob/main/doc/src/atomvm-tooling.md?plain=1#L352
https://github.com/atomvm/AtomVM/blob/main/doc/src/atomvm-tooling.md?plain=1#L355
https://github.com/atomvm/AtomVM/blob/main/doc/src/atomvm-tooling.md?plain=1#L372

the example mentions `atomvm.esp32.flash`:
https://github.com/atomvm/AtomVM/blob/main/doc/src/atomvm-tooling.md?plain=1#L361

I think the example is correct since the mix task defined (here)[https://github.com/atomvm/exatomvm/blob/main/lib/mix/tasks/esp32_flash.ex#L1] is called `Mix.Tasks.Atomvm.Esp32.Flash`

a-maze-d, 2025-05-25:
ok, I can raise. PR and also fix yhe duplicate note (further down) if you like

UncleGrumpy, 2025-05-25:
That would be great if you want to take care of that. The fix should be based on the release-0.6 branch so that it will be corrected in the next update. The fixes merged into release-0.6 branch will be forwarded to main periodically, so you don’t need to worry about updating the main branch.

Don’t forget to sign your commit, and if you include “Closes #1677” (without quotes) after the description this issue will be closed automatically after your PR gets merged and forwarded to the main branch.

commit 8faa85b8 (2025-05-26): Fixed minor errors in the tooling documentation

* mix.esp32.flash --> atomvm.esp32.flash
*  mix.stm32.flash --> atomvm.stm32.flash
* removed duplicate note block (kept the {tip} version
* instead of calling it `target` called it `mix ...  task`

Closes #1677
closed 2025-05-29 by bettio