atomvm/AtomVM #1683: [Help Wanted] How to implement support for STM32H743VIT6 (or any other STM32H7)?

dg1sbg, 2025-05-31:
Hello all:

We have developed a Software Defined Radio DSP board that employs an STM32H743VIT6 MCU as a board controller and audio and control data gateway. Having heard of the capabilities the AtomVM now has, we would like to contribute to implement support for this and also other STM32H7 MCUs.

Question is: How to do that?

Is there a plartform support implementation guide somewhere - or, even better, someone to chat with that could provide some guidance?

Thanks for feedback.

Best,
    Frank

UncleGrumpy, 2025-05-31:
We would be thrilled to have some help improving our stm32 support. Presently the STM32 platform is built with libopencm3. As long as the device is supported by libopencm3 already you should just be able to build AtomVM for that target.

We don’t really have a porting guide, but would certainly be happy to chat in our telegram community:
https://t.me/atomvm

dg1sbg, 2025-05-31:
It would be a pleasure to be contributing, @UncleGrumpy . Just looked into libopencm3 and found out that the STM32H7 family should be supported. Now, I think I missed how to specify the device correctly on the copmpile command:

In `AtomVM/src/platforms/stm32/src`, I do:

`cmake -DCMAKE_TOOLCHAIN_FILE=../cmake/arm-toolchain.cmake -DLIBOPENCM3_DIR=/opt/libopencm3 -DDEVICE=stm32h743vi6 ..`

and I get:

`----------------------------------------
-- Device      : stm32h743vi6
--------Device Configuration Info-------
-- Clock Hz    : Error! Unsupported device stm32h743vi6.

-- Flash Size  : Error! Unsupported device stm32h743vi6.

-- Console     : CONSOLE_2
-----------Device Linker Info-----------
-- Family      : stm32h7
-- Sub-family  : stm32h7xxxi
-- CPU         : cortex-m7
-- FPU         : hard-fpv5-d16
-- CPP Flags   : -DSTM32H7 -DSTM32H7XXXI -DSTM32H743VI6
-- Linker Defs : -DSTM32H7 -DSTM32H7XXXI -DSTM32H743VI6 -D_RAM=512K -D_RAM2=128K -D_RAM3=128K -D_RAM4=32K -D_RAM5=64K -D_CCM=128K -D_ROM=1M -D_ROM2=1M -D_ROM_OFF=0x08000000 -D_ROM2_OFF=0x08100000 -D_RAM_OFF=0x24000000 -D_RAM2_OFF=0x30000000 -D_RAM3_OFF=0x30020000 -D_RAM4_OFF=0x30040000 -D_RAM5_OFF=0x38000000 -D_CCM_OFF=0x20000000
CMake Warning at cmake/libopencm3.cmake:111 (message):
  No match for the FPU flags
Call Stack (most recent call first):
  CMakeLists.txt:109 (include)

-----------Target Specific Info---------
-- Generated Linker File   : /var/data/devtools/atomvm/src/AtomVM/src/platforms/stm32/src/generated.stm32h743vi6.ld
-- Linker Flags            :  -specs=nosys.specs -nostartfiles -Wl,--undefined,_printf_float -Wl,--undefined,_scanf_float -T/var/data/devtools/atomvm/src/AtomVM/src/platforms/stm32/src/generated.stm32h743vi6.ld -mcpu=cortex-m7 -mthumb -mthumb-interwork
-- Target Specific Flags   : -DSTM32H7 -DSTM32H7XXXI -DSTM32H743VI6 -mcpu=cortex-m7 -mthumb -mthumb-interwork
------------Output Locations------------
-- BIN: /var/data/devtools/atomvm/src/AtomVM/src/platforms/stm32/src/AtomVM-stm32h743vi6.bin
-- ELF: /var/data/devtools/atomvm/src/AtomVM/src/platforms/stm32/src/AtomVM-stm32h743vi6.elf
-- Configuring done (1.9s)
-- Generating done (0.4s)
-- Build files have been written to: /var/data/devtools/atomvm/src/AtomVM/src/platforms/stm32/src`

Now, what worries me are the two lines 
`-- Clock Hz    : Error! Unsupported device stm32h743vi6.

-- Flash Size  : Error! Unsupported device stm32h743vi6.
`
... and I obviously need to fix this ...

Any ideas or pointers?

Best,
   Frank

Sorry - I can't use Telegram.

UncleGrumpy, 2025-05-31:
Yes, sorry, I forgot that I have not added the H7 devices to our configuration system. I do not have access to any to test with. So what needs to happen next is we need the device info added to:

https://github.com/atomvm/AtomVM/blob/main/src/platforms/stm32/tools/device_config.hrl

You should also check and see if the flash size and ram size are already covered in avm_devcfg.h of if they need to be added:

https://github.com/atomvm/AtomVM/blob/main/src/platforms/stm32/src/lib/avm_devcfg.h

If you don't/can't use Telegram we also have a lightly used discord server if that is an option for you:
https://discord.gg/5SJnuPhv

UncleGrumpy, 2025-05-31:
The definition in device_config.hrl should match what libopencm3 defines for the device.

UncleGrumpy, 2025-05-31:
Before you invest too much time I should make sure you understand that the STM32 platform has the least support of all at the moment. It currently only supports a GPIO driver as far as peripherals go. We would like to expand this, but my time has been limited and the libopencm3 documentation is quite sparse. But I am certainly willing to help improve this as much as possible.

dg1sbg, 2025-06-01:
Thanks for your offer to help expanding STM32 support. I need to have Ethernet and also SPI support in addition to the already existing GPIO functionality.

I see that the device configuration is quite fixed - the definitions in  `https://github.com/atomvm/AtomVM/blob/main/src/platforms/stm32/src/lib/avm_devcfg.h` can't be changed based on RCC, PLL, ... and I/O config settings - and it is just impossible to implement all the configuration combinations. I would rather want a config file like the .ld linker scripot having all the config vars set to the specific settings and rather not hard-code stuff. Or am I missing something?

Thx again for feedback.

// Frank

dg1sbg, 2025-06-01:
On my quest to understand what is going on I ran cmake with the -LAH option. That provides the following output:

```
$ cmake -LAH -DCMAKE_TOOLCHAIN_FILE=../cmake/arm-toolchain.cmake -DLIBOPENCM3_DIR=/opt/libopencm3 -DDEVICE=stm32h743vi6 ..
----------------------------------------
-- Device      : stm32h743vi6
--------Device Configuration Info-------
-- Clock Hz    : Error! Unsupported device stm32h743vi6.

-- Flash Size  : Error! Unsupported device stm32h743vi6.

-- Console     : CONSOLE_2
-----------Device Linker Info-----------
-- Family      : stm32h7
-- Sub-family  : stm32h7xxxi
-- CPU         : cortex-m7
-- FPU         : hard-fpv5-d16
-- CPP Flags   : -DSTM32H7 -DSTM32H7XXXI -DSTM32H743VI6
-- Linker Defs : -DSTM32H7 -DSTM32H7XXXI -DSTM32H743VI6 -D_RAM=512K -D_RAM2=128K -D_RAM3=128K -D_RAM4=32K -D_RAM5=64K -D_CCM=128K -D_ROM=1M -D_ROM2=1M -D_ROM_OFF=0x08000000 -D_ROM2_OFF=0x08100000 -D_RAM_OFF=0x24000000 -D_RAM2_OFF=0x30000000 -D_RAM3_OFF=0x30020000 -D_RAM4_OFF=0x30040000 -D_RAM5_OFF=0x38000000 -D_CCM_OFF=0x20000000
CMake Warning at cmake/libopencm3.cmake:111 (message):
  No match for the FPU flags
Call Stack (most recent call first):
  CMakeLists.txt:109 (include)

-----------Target Specific Info---------
-- Generated Linker File   : /var/data/devtools/atomvm/src/AtomVM/src/platforms/stm32/src/generated.stm32h743vi6.ld
-- Linker Flags            :  -specs=nosys.specs -nostartfiles -Wl,--undefined,_printf_float -Wl,--undefined,_scanf_float -T/var/data/devtools/atomvm/src/AtomVM/src/platforms/stm32/src/generated.stm32h743vi6.ld -mcpu=cortex-m7 -mthumb -mthumb-interwork
-- Target Specific Flags   : -DSTM32H7 -DSTM32H7XXXI -DSTM32H743VI6 -mcpu=cortex-m7 -mthumb -mthumb-interwork
------------Output Locations------------
-- BIN: /var/data/devtools/atomvm/src/AtomVM/src/platforms/stm32/src/AtomVM-stm32h743vi6.bin
-- ELF: /var/data/devtools/atomvm/src/AtomVM/src/platforms/stm32/src/AtomVM-stm32h743vi6.elf
-- Configuring done (1.9s)
-- Generating done (0.3s)
-- Build files have been written to: /var/data/devtools/atomvm/src/AtomVM/src/platforms/stm32/src
-- Cache values
// Path to a program.
ARM_CC:FILEPATH=/opt/arm-toolchain-stm32/bin/arm-none-eabi-gcc

// Path to a program.
ARM_CXX:FILEPATH=/opt/arm-toolchain-stm32/bin/arm-none-eabi-g++

// Path to a program.
ARM_OBJCOPY:FILEPATH=/opt/arm-toolchain-stm32/bin/arm-none-eabi-objcopy

// Path to a program.
ARM_SIZE:FILEPATH=/opt/arm-toolchain-stm32/bin/arm-none-eabi-size

// AtomVM system console uart
AVM_CFG_CONSOLE:STRING=CONSOLE_2

// Reboot when application exits with non 'ok' return
AVM_CONFIG_REBOOT_ON_NOT_OK:BOOL=OFF

// Create stacktraces
AVM_CREATE_STACKTRACES:BOOL=ON

// Disable GPIO nifs (input and output)
AVM_DISABLE_GPIO_NIFS:BOOL=OFF

// Disable GPIO 'port' driver (input, output, and interrupts)
AVM_DISABLE_GPIO_PORT_DRIVER:BOOL=OFF

// Use color log output
AVM_ENABLE_LOG_COLOR:BOOL=OFF

// Include source and line info for all enbled levels
AVM_ENABLE_LOG_LINES:BOOL=OFF

// Disable log output
AVM_LOG_DISABLE:BOOL=OFF
[... 190 lines cut ...]

In the above output the  following seems to be ok:

```
ARM_CC:FILEPATH=/opt/arm-toolchain-stm32/bin/arm-none-eabi-gcc
ARM_CXX:FILEPATH=/opt/arm-toolchain-stm32/bin/arm-none-eabi-g++
ARM_OBJCOPY:FILEPATH=/opt/arm-toolchain-stm32/bin/arm-none-eabi-objcopy
ARM_SIZE:FILEPATH=/opt/arm-toolchain-stm32/bin/arm-none-eabi-size
```

but these might be problematic:

```
CMAKE_C_COMPILER:FILEPATH=/usr/bin/cc
CMAKE_C_COMPILER_AR:FILEPATH=/usr/bin/gcc-ar-13
CMAKE_C_COMPILER_RANLIB:FILEPATH=/usr/bin/gcc-ranlib-13

CMAKE_LINKER:FILEPATH=/usr/bin/ld
CMAKE_MAKE_PROGRAM:FILEPATH=/usr/bin/gmake
 ...
```

I am not a cmake user normally, so I am trying to get hold of something ...

// Frank

UncleGrumpy, 2025-06-01:
Indeed. Those are problematic. CMAKE_CXX_COMPILER_AR, CMAKE_CXX_COMPILER_RANLIB and CMAKE_LINKER, should be using the arm toolchain versions. CMAKE_C_COMPILER should not be set at all. I have not build with gmake so I don't know if that would cause any issues. Do you have a compiler already set in your environment possibly?

UncleGrumpy, 2025-06-01:
Cmake will pick up environment settings, so if CC or CMAKE_C_COMPILER are set in your environment that may be overriding the AR, RANLIB and LINKER settings.

UncleGrumpy, 2025-06-01:
> Thanks for your offer to help expanding STM32 support. I need to have Ethernet and also SPI support in addition to the already existing GPIO functionality.

I do have an SPI driver on my todo list. I had put that on hold hoping to rebase the platform on Zephyr, but that turned into a much bigger challenge than I first expected. Adding an SPI driver should not be very difficult. I have not looked into Ethernet support yet, but that is also a feature I would like to add, but I expect will be a much bigger job.

> I see that the device configuration is quite fixed - the definitions in https://github.com/atomvm/AtomVM/blob/main/src/platforms/stm32/src/lib/avm_devcfg.h can't be changed based on RCC, PLL, ... [...]

I do not have a ton of experience with STM32, so this is a bit of a naive implementation. I am much more familiar with the ESP32 platform. We would welcome any improvements. I went with maximum clock speed for a given device mainly for simplicity, we would need to add validation to make sure that the clock options supplied were supported by the device and would not break console output. I did not offer all of the alternative function console options again for simplicity, I had assumed that if the default console was not suitable for a given build one of the other 7 options would be acceptable. None of this is set in stone. We could offer more configuration options, I am certainly open to suggestions.

UncleGrumpy, 2025-06-02:
It just occurred to me I did not ask if you made sure to run `make` inside the libopencm3 directory before trying to use it. If not that could be part of your troubles.

I have just pushed the missing configuration info to a branch on my fork of AtomVM:

https://github.com/UncleGrumpy/AtomVM/tree/stm32h7

Simplest is probably to add my fork as a remote and check out the branch:
```
$ git add remote winford https://github.com/UncleGrumpy/AtomVM
$ git fetch --all
$ git checkout winford/stm32h7
```

I don't know if the clock setting I added for the h7 make sense or work with the uart settings, I can't test that... If they don't work, or there is a better choice please let me know.

dg1sbg, 2025-06-02:
> Cmake will pick up environment settings, so if CC or CMAKE_C_COMPILER are set in your environment that may be overriding the AR, RANLIB and LINKER settings.

No, I made sure that I don't get any CMAKE stuff before running the cmake command. I also don't explicitely set CC as an env var. 

See more info on this at [my comment in issue 1784](https://github.com/atomvm/AtomVM/issues/1684#issuecomment-2927830307).

dg1sbg, 2025-06-02:
> The clock setting I used there are definitely wrong. It looks like the H7 series uses a completely different configuration method than the f4/f7 series chips, and unfortunately there are no examples at all for the h7 in the libopencm3-examples repo for me to use as a reference.

Yep, I am used to do the H7 config using the STM32CubeIDE from ST. I would like to be able to use this IDE to graphically config pin assignments, port and clock settings. The IDE generates C or C++ code and include files. Food for thought. Thinking ...

UncleGrumpy, 2025-06-02:
> Yep, I am used to do the H7 config using the STM32CubeIDE from ST. I would like to be able to use this IDE to graphically config pin assignments, port and clock settings. The IDE generates C or C++ code and include files. Food for thought. Thinking ...

That would require a complete rewrite of the stm32 port to use STM32CubeIDE libraries. That would dramatically increase the complexity of the builds and CI workflows, and the size of the resulting builds. Libopencm3 is extremely lean and efficient, using it allows us to support devices with only 512k of flash, like the black pill v2, only just barely. I think the builds on main branch are slightly too large for these devices at the moment, but there are a few open PR that will bring the size of the build back down to be able to continue to support these smaller flash sizes.

Since the only pins that we set in the VM itself are those used for the console UART I don’t see a lot of advantage in the GUI pin configuration. Any other pins would be set in the BEAM application at runtime.

I do think we need to expose the clock configuration so that it can be adjusted with a cmake option. We might be able to expose this to the Erlang layer so that the clock could be adjusted in the application at runtime, but I have not looked into this at all.

UncleGrumpy, 2025-06-02:
The clock setting in my branch are almost certainly wrong, I was just guessing, but it looks like the clock configuration for the H7 is slightly different. I can’t be sure since there are no examples at all for the H7 in the libopencm3-examples repo.

UncleGrumpy, 2025-06-02:
>CMAKE_MAKE_PROGRAM:FILEPATH=/usr/bin/gmake

That part is correct, I get the same result there, but the other paths should be using the ARM toolchain variants. I am really confused by this. I don’t understand why cmake is setting these to the standard gcc versions.

UncleGrumpy, 2025-06-02:
I had a few typos I did not catch last night, those are fixed and I attempted to get a working clock configuration for the H7, I still have doubts that it is and correct. I will try to look into this, but it is hard without being able to test on hardware.

UncleGrumpy, 2025-06-06:
>Extend libopencm3 support all H7 ports as stated earlier. Only ETH support seems to be missing completely - but I haven't looked cosely yet.

Actually ethernet support is included in libopencm3 here:
https://github.com/libopencm3/libopencm3/tree/master/include/libopencm3/ethernet

I found a usage example here:
https://github.com/berndoJ/stm32f107_ethernet_test

That example is written for stm32f107, but the ethernet driver itself is a portable platform agnostic implementation that works for all of the libopencm3 devices... not just the stm32 families.

Yes that is the right approach. For examples of exposing then nifs/ports to the Erlang layer you can look at the existing GPIO driver. It has read and write implementations of nifs and ports, the interrupts are currently only available as a port, but we could implement them as resource based nifs as well.