atomvm/AtomVM #1737: strange behavior (spinnoff from#1734)

elsbiet, 2025-07-03:
i'm closing #1734 because i feel this is the same issue.

hw = esp32-s3,
Atomvm = binary download ( https://github.com/atomvm/AtomVM/releases/download/v0.6.6/AtomVM-esp32s3-v0.6.6.img )

`-module(main).

-export([start/0]).

start() ->
%   L = <<$t,$e>>,                         % runs fine
    L = <<1,2>>,                               % crashes
    T = fun T(2) ->
                ok
                ;
            T(N) ->
                <<_:N/binary, X:1/binary, _/binary>> = L,
                io:format("G=~p~n", [X]),

                T(N+1)
        end,
    T(0)
    .
`
with L = <<$t,$e>> runs fine
`   ###########################################################

       ###    ########  #######  ##     ## ##     ## ##     ## 
      ## ##      ##    ##     ## ###   ### ##     ## ###   ### 
     ##   ##     ##    ##     ## #### #### ##     ## #### #### 
    ##     ##    ##    ##     ## ## ### ## ##     ## ## ### ## 
    #########    ##    ##     ## ##     ##  ##   ##  ##     ## 
    ##     ##    ##    ##     ## ##     ##   ## ##   ##     ## 
    ##     ##    ##     #######  ##     ##    ###    ##     ## 

    ###########################################################

I (671) AtomVM: Starting AtomVM revision 0.6.6
I (681) sys: Loaded BEAM partition boot.avm at address 0x1d0000 (size=262144 bytes)
I (701) network_driver: Initialized network interface
I (701) network_driver: Created default event loop
I (711) AtomVM: Found startup beam esp32init.beam
I (711) AtomVM: Starting esp32init.beam...
---
AtomVM init.
I (721) sys: Loaded BEAM partition main.avm at address 0x210000 (size=1048576 bytes)
Starting application...
G=<<"t">>
G=<<"e">>
AtomVM finished with return value: ok
I (761) AtomVM: AtomVM application terminated.  Going to sleep forever ...
`

-----

with  L = <<1,2>> crashes:
`I (603) main_task: Started on CPU0
I (633) main_task: Calling app_main()

    ###########################################################

       ###    ########  #######  ##     ## ##     ## ##     ## 
      ## ##      ##    ##     ## ###   ### ##     ## ###   ### 
     ##   ##     ##    ##     ## #### #### ##     ## #### #### 
    ##     ##    ##    ##     ## ## ### ## ##     ## ## ### ## 
    #########    ##    ##     ## ##     ##  ##   ##  ##     ## 
    ##     ##    ##    ##     ## ##     ##   ## ##   ##     ## 
    ##     ##    ##     #######  ##     ##    ###    ##     ## 

    ###########################################################

I (673) AtomVM: Starting AtomVM revision 0.6.6
I (683) sys: Loaded BEAM partition boot.avm at address 0x1d0000 (size=262144 bytes)
I (703) network_driver: Initialized network interface
I (703) network_driver: Created default event loop
I (713) AtomVM: Found startup beam esp32init.beam
I (713) AtomVM: Starting esp32init.beam...
---
AtomVM init.
I (723) sys: Loaded BEAM partition main.avm at address 0x210000 (size=1048576 bytes)
Starting application...
G=<<1>>
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42041221  PS      : 0x00060930  A0      : 0x82052ce1  A1      : 0x3fca8ed0  
A2      : 0x3c1a0126  A3      : 0x0000000a  A4      : 0x3fcb09fa  A5      : 0x3c1a011e  
A6      : 0x3c1a0124  A7      : 0x00000008  A8      : 0x00000001  A9      : 0x00000008  
A10     : 0x3fcb09f8  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x3c1a0122  
A14     : 0x00000000  A15     : 0x3fcb282c  SAR     : 0x00000009  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  

Backtrace: 0x4204121e:0x3fca8ed0 0x42052cde:0x3fca8f80 0x4201077d:0x3fca8fa0 0x421211e3:0x3fca8fe0
`

rem.: the behavior is identical for AtomVM revision 0.7.0-dev+git.515ec037 (self-compiled)

elsbiet, 2025-07-03:
additional info:

the pgm crashes for

L = <<0,1>>
L = <<1,2>>
..
L = <<7, 8>>

and runs fine for

L = <<8,9>>
..

petermm, 2025-07-03:
can you format the code blocks? https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks

it's difficult to understand as is..

UncleGrumpy, 2025-07-04:
I'm not sure what is causing the crash on your ESP32S3, I just tested you example on and ESP32S3 with a freshly downloaded and flashed release v0.6.6 image and you example works as expected:
```

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x11f0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd04
load:0x403cb700,len:0x2e50
entry 0x403c8918
I (27) boot: ESP-IDF v5.4.1 2nd stage bootloader
I (27) boot: compile time Jun 23 2025 22:02:09
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.1
I (30) boot: efuse block revision: v1.2
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 001c0000
I (78) boot:  3 boot.avm         RF data          01 01 001d0000 00040000
I (84) boot:  4 main.avm         RF data          01 01 00210000 00100000
I (91) boot: End of partition table
I (94) esp_image: segment 0: paddr=00010020 vaddr=3c130020 size=2a0dch (172252) map
I (132) esp_image: segment 1: paddr=0003a104 vaddr=3fc9c900 size=044a8h ( 17576) load
I (136) esp_image: segment 2: paddr=0003e5b4 vaddr=40374000 size=01a64h (  6756) load
I (138) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=121d74h (1187188) map
I (353) esp_image: segment 4: paddr=00161d9c vaddr=40375a64 size=16db8h ( 93624) load
I (373) esp_image: segment 5: paddr=00178b5c vaddr=600fe000 size=00100h (   256) load
I (374) esp_image: segment 6: paddr=00178c64 vaddr=600fe100 size=00020h (    32) load
I (386) boot: Loaded app from partition at offset 0x10000
I (387) boot: Disabling RNG early entropy source...
I (398) cpu_start: Multicore app
I (408) cpu_start: Pro cpu start user code
I (408) cpu_start: cpu freq: 160000000 Hz
I (408) app_init: Application information:
I (408) app_init: Project name:     atomvm-esp32
I (412) app_init: App version:      v0.6.6-dirty
I (417) app_init: Compile time:     Jun 23 2025 22:02:03
I (422) app_init: ELF file SHA256:  2afe9b139...
I (426) app_init: ESP-IDF:          v5.4.1
I (430) efuse_init: Min chip rev:     v0.0
I (434) efuse_init: Max chip rev:     v0.99 
I (438) efuse_init: Chip rev:         v0.1
I (442) heap_init: Initializing. RAM available for dynamic allocation:
I (448) heap_init: At 3FCA54D0 len 00044240 (272 KiB): RAM
I (453) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (458) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (463) heap_init: At 600FE120 len 00001EC8 (7 KiB): RTCRAM
I (470) spi_flash: detected chip: gd
I (472) spi_flash: flash io: dio
W (475) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
W (488) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (497) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (503) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (510) main_task: Started on CPU0
I (520) main_task: Calling app_main()

    ###########################################################

       ###    ########  #######  ##     ## ##     ## ##     ## 
      ## ##      ##    ##     ## ###   ### ##     ## ###   ### 
     ##   ##     ##    ##     ## #### #### ##     ## #### #### 
    ##     ##    ##    ##     ## ## ### ## ##     ## ## ### ## 
    #########    ##    ##     ## ##     ##  ##   ##  ##     ## 
    ##     ##    ##    ##     ## ##     ##   ## ##   ##     ## 
    ##     ##    ##     #######  ##     ##    ###    ##     ## 

    ###########################################################

I (570) AtomVM: Starting AtomVM revision 0.6.6
I (570) sys: Loaded BEAM partition boot.avm at address 0x1d0000 (size=262144 bytes)
I (590) network_driver: Initialized network interface
I (590) network_driver: Created default event loop
I (600) AtomVM: Found startup beam esp32init.beam
I (600) AtomVM: Starting esp32init.beam...
---
AtomVM init.
I (610) sys: Loaded BEAM partition main.avm at address 0x210000 (size=1048576 bytes)
Starting application...
G=<<0>>
G=<<1>>
AtomVM finished with return value: ok
I (630) AtomVM: AtomVM application terminated.  Going to sleep forever ...
```

Did you erase the flash on your esp32s3 before flashing the updated image?

`esptool.py --chip auto --baud 921600 erase_flash`

That is the only thing I can think of that might be leading to the crash you are seeing, there may be old data being read, either when loading the VM or in the application partition.

elsbiet, 2025-07-05:
1. `esptool.py --chip esp32s3 --baud 921600 -p /dev/ttyACM1 erase_flash`

2. `wget -S https://github.com/atomvm/AtomVM/releases/download/v0.6.6/AtomVM-esp32s3-v0.6.6.img`

3. `sha256sum -c AtomVM-esp32s3-v0.6.6.img.sha256`
`AtomVM-esp32s3-v0.6.6.img: OK`

4.``` esptool.py \
--chip auto \
--port /dev/ttyACM1 --baud 921600 \
--before default_reset --after hard_reset \
write_flash -u \
--flash_mode dio --flash_freq 40m --flash_size detect \
0x000 \
./AtomVM-esp32s3-v0.6.6.img ```

5. program: 
```
-module(main).

-export([start/0]).

start() ->
    L = <<0,1>>,
    T = fun T(2) ->
                ok
                ;
            T(N) ->
                <<_:N/binary, X:1/binary, _/binary>> = L,
                io:format("G=~p~n", [X]),

                T(N+1)
        end,
    T(0)
    .
```

6. ```cd .. && \
    rebar3 atomvm packbeam  && \
    rebar3 atomvm esp32_flash --esptool /home/splge/.espressif/python_env/idf5.4_py3.13_env/bin/esptool.py --chip auto --port /dev/ttyACM1 --baud 921600 --offset 0x210000```

result:

``` ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403758fc
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce2810,len:0x11f0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd04
load:0x403cb700,len:0x2e50
entry 0x403c8918
I (31) boot: ESP-IDF v5.4.1 2nd stage bootloader
I (31) boot: compile time Jun 23 2025 22:02:09
I (31) boot: Multicore bootloader
I (31) boot: chip revision: v0.2
I (34) boot: efuse block revision: v1.3
I (38) boot.esp32s3: Boot SPI Speed : 40MHz
I (42) boot.esp32s3: SPI Mode       : DIO
I (45) boot.esp32s3: SPI Flash Size : 16MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 001c0000
I (82) boot:  3 boot.avm         RF data          01 01 001d0000 00040000
I (89) boot:  4 main.avm         RF data          01 01 00210000 00100000
[... 53 lines cut ...]
---
AtomVM init.
I (723) sys: Loaded BEAM partition main.avm at address 0x210000 (size=1048576 bytes)
Starting application...
G=<<0>>
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42041221  PS      : 0x00060930  A0      : 0x82052ce1  A1      : 0x3fca8ed0  
A2      : 0x3c1a0126  A3      : 0x0000000a  A4      : 0x3fcb09fa  A5      : 0x3c1a011e  
A6      : 0x3c1a0124  A7      : 0x00000008  A8      : 0x00000001  A9      : 0x00000008  
A10     : 0x3fcb09f8  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x3c1a0122  
A14     : 0x00000000  A15     : 0x3fcb282c  SAR     : 0x00000009  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  

Backtrace: 0x4204121e:0x3fca8ed0 0x42052cde:0x3fca8f80 0x4201077d:0x3fca8fa0 0x421211e3:0x3fca8fe0

```

maybe a difference of the erlang runtime on the develpment pc - mine is 
```erl
Erlang/OTP 27 [erts-15.2.2] [source] [64-bit] [smp:16:16] [ds:16:16:10] [async-threads:1] [jit:ns]

Eshell V15.2.2 (press Ctrl+G to abort, type help(). for help)
```

UncleGrumpy, 2025-07-09:
That is absolutely true for a source build, but the release images do contain the library (esp32boot.avm) partition. Flashing a release image should work without the failure you encountered.

elsbiet, 2025-07-22:
I am happy to announce that the problem seems to have vanished in AtomVM revision 0.7.0-dev+git.58628068.

thanks a lot for fixing.