atomvm/AtomVM #1781: Documentation: AtomVM Tooling

elsbiet, 2025-07-28:
[atomvm_rebar3_plugin](https://doc.atomvm.org/main/atomvm-tooling.html)

when upgrading otp-27  -> otp-28 it seems to be necessary to

1. compile/install (at least) rebar 3.25.0
2. `rm -rf ~/.cache/rebar3`
3. `rm -rf ${Project}/_build ${Project}/rebar.lock`

if one of step2 or step 3 is ommitted, then (at least in my development environment) rebar3 will fail with

```
===> An error occurred in the packbeam task.  Class=error Error=data_error Stacktrace=[{zlib,
                                                                                                   inflate_nif,
                                                                                                   [#Ref<0.1927247248.1290928141.224031>,
                                                                                                    8192,
                                                                                                    16384,
                                                                                                    0],
                                                                                                   []},
                                                                                                  {zlib,
                                                                                                   dequeue_all_chunks_1,
                                                                                                   3,
                                                                                                   []},
...
```

UncleGrumpy, 2025-07-28:
Indeed, this is not an AtomVM issue, it will happen for OTP apps as well, but we should add a mention of this in the documentation.

elsbiet, 2025-07-31:
going to try pr - i'm inexperienced ..