atomvm/AtomVM #1782: AtomVm-0.7 (main): possible memory leak in network / distribution stack.

elsbiet, 2025-07-29:
i created a little service

```
-module(oom).
-behaviour(gen_server).

-include_lib("kernel/include/logger.hrl").

-export([start_link/1]).
-export([init/1, handle_call/3, handle_cast/2, handle_info/2, terminate/2]).

-record(state, {
          call_count = <<>>
         }
       ).

start_link(Arg) ->
    gen_server:start_link({local, ?MODULE}, ?MODULE, Arg, [])
    .

init(Arg) ->
    {ok, #state{call_count=Arg}}
    .

handle_call(Request, _From, #state{call_count = Callcount} = State) ->
    ?LOG_DEBUG("Request=~p", [Request]),
    {reply, Callcount, State#state{call_count = Callcount + 1}}
    .

handle_cast(_E, State) ->
    {noreply, State}
    .

handle_info(E, State) ->
    ?LOG_DEBUG("unexpected: ~p", [E]),
    {noreply, State}
    .

terminate(_Reason, _Tab) ->
    ?LOG_DEBUG("done"),
    ok.
```
and started it on an esp32-s3. 

than i called it repeatedly from a linux node.
```-module(t).

-export([t/1]).

t(0) -> ok;
t(N) ->
    Response = gen_server:call({oom, 'atomvm@host.domain.tld'}, N),
    io:format("Response=~p~n", [Response]),
    timer:sleep(100),
    t(N-1).
'''

after 1022 cycles the service crashed leaving the following info

```1970-01-03T00:47:41.469Z [debug] <0.14.0> oom:handle_call/3 (/home/xxxx/ATOM/MyAtomVM/erlang/esp32/oom/src/oom.erl:23) Request=8981
1970-01-03T00:47:41.716Z [debug] <0.14.0> oom:handle_call/3 (/home/xxxx/ATOM/MyAtomVM/erlang/esp32/oom/src/oom.erl:23) Request=8980
1970-01-03T00:47:41.965Z [debug] <0.14.0> oom:handle_call/3 (/home/xxxx/ATOM/MyAtomVM/erlang/esp32/oom/src/oom.erl:23) Request=8979
WARNING: Unable to allocate heap space for raw stacktrace
CRASH
======
pid: <0.14.0>

Stacktrace:
out_of_memory

cp: #CP<module: 4, label: 100, offset: 38>

x[0]: error
x[1]: out_of_memory
x[2]: out_of_memory

Stack
-----

[]
[]
[]
{state,1022}
{<879.276.0>,[alias|#Ref<879.0.35331.4167739403.1000407044.59928>]}
{state,oom,oom,{state,1022}}
<0.1.0>
#CP<module: 4, label: 140, offset: 0>

Mailbox
-------

Monitors
--------
link to <0.1.0>
monitored by resource 0x3fcc6d54 ref=54
monitored by resource 0x3fcc6d54 ref=63
monitored by resource 0x3fcc6d54 ref=70
monitored by resource 0x3fcc6d54 ref=7193
...
~1000 others lines like these
...
monitored by resource 0x3fcc6d54 ref=7200
monitored by resource 0x3fcc6d54 ref=7207
monitored by resource 0x3fcc6d54 ref=7214
monitored by resource 0x3fcc6d54 ref=7221
monitored by resource 0x3fcc6d54 ref=7228

Context memory info
-------------------
context_size = 3996
context_avail_free_memory = 1
heap_size = 914
total_heap_size = 939
stack_size = 8
message_queue_len = 0

Global memory info
------------------
process_count = 14
ports_count = 1
atoms_count = 897
refc_binary_total_size = 344

**End Of Crash Report**
Cannot handle out of memory.

abort() was called at PC 0x4202799f on core 0
0x4202799f: context_monitors_handle_terminate at /home/xxxx/ATOM/dist/AtomVM/src/libAtomVM/context.c:641

Backtrace: 0x40375f29:0x3fcab7e0 0x4037f98d:0x3fcab800 0x40388bf5:0x3fcab820 0x4202799f:0x3fcab890 0x420286d1:0x3fcab8b0 0x420848ad:0x3fcab8e0 0x42057cf5:0x3fcab900 0x42057d9d:0x3fcab9b0 0x4205c7e6:0x3fcab9e0 0x42011e31:0x3fcaba00 0x4212ea00:0x3f
0x40375f29: panic_abort at /home/xxxx/esp/esp-idf/components/esp_system/panic.c:468

0x4037f98d: esp_system_abort at /home/xxxx/esp/esp-idf/components/esp_system/port/esp_system_chip.c:87

0x40388bf5: abort at /home/xxxx/esp/esp-idf/components/newlib/src/abort.c:38

0x4202799f: context_monitors_handle_terminate at /home/xxxx/ATOM/dist/AtomVM/src/libAtomVM/context.c:641

0x420286d1: context_destroy at /home/xxxx/ATOM/dist/AtomVM/src/libAtomVM/context.c:219

0x420848ad: scheduler_terminate at /home/xxxx/ATOM/dist/AtomVM/src/libAtomVM/scheduler.c:432

0x42057cf5: scheduler_entry_point at /home/xxxx/ATOM/dist/AtomVM/src/libAtomVM/opcodesswitch.h:7221

0x42057d9d: context_execute_loop at /home/xxxx/ATOM/dist/AtomVM/src/libAtomVM/opcodesswitch.h:1750

0x4205c7e6: globalcontext_run at /home/xxxx/ATOM/dist/AtomVM/src/libAtomVM/globalcontext.c:739

0x42011e31: app_main at /home/xxxx/ATOM/dist/AtomVM/src/platforms/esp32/main/main.c:126 (discriminator 1)

0x4212ea00: main_task at /home/xxxx/esp/esp-idf/components/freertos/app_startup.c:208

0x403804f5: vPortTaskWrapper at /home/xxxx/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

this does not happen when the service is called locally on the esp.
```
    Do = fun Do(0) ->
                 ok;
             Do(N) ->
                 Response = gen_server:call(oom, N),
                 io:format("Response=~p~n", [Response]),
                 Do(N - 1)
         end,

    Do(20000),
```
runs until N = 0.

all programs were compiled with otp-28 erlc.

pguyot, 2025-08-08:
Thank you for the report.

Monitors created by disterl call from BEAM should be released and aren't.

The monitors belong to a single resource, probably the dist connection.

elsbiet, 2025-08-13:
additional info: after i replaced

`Response = gen_server:call({oom, 'atomvm@host.domain.tld'}, N),`

by

`ok = gen_server:cast({oom, 'atomvm@host.domain.tld'}, N),`

my testprogram survived 250000 cycles which suggests to me that the leak is reply-related.

pguyot, 2025-08-13:
Can you please try this?

#1800 
pguyot:w33/fix-disterl-demonitor-leak

I will have to amend to fix the commit message when I’m back to a keyboard.

elsbiet, 2025-08-14:
@pguyot , @bettio  I am happy to report that my test has just completed 250,000 cycles without any problems. I therefore assume that the problem has been resolved and request that the PR be merged.

Many thanks, E

PR #1790 (2025-08-10): Fix leak with disterl monitor by name
Also add `monitored_by` in `process_info/2` to test this fix

Fix #1782 

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
commit 71a16108 (2025-09-03): Fix leak in disterl demonitor

Fixes #1782