atomvm/AtomVM #1787: diesterl gen_server:call form atomvm fails

elsbiet, 2025-08-06:
from esp32
```
 {server, remote_node} ! <<"hello world from atomvm">>
```
works as expected but
```
gen_server:call({server, remote_node}, <<"hello world from atomvm">>)
````
throws an exception
```
.. AtomVM/libs/estdlib/src/gen_server.erl"},{line,434}]}]
```

pguyot, 2025-08-06:
Could you please post the full error log? It probably can easily be reproduced but I'm afk right now.

I guess this is from AtomVM to BEAM? I.e. the gen_server process runs on a BEAM node? Which OTP version?

We might miss a bit of disterl implementation or a gen_server protocol compatibility with BEAM.

elsbiet, 2025-08-06:
1. no more info than this:
```
CRASH 
======
pid: <0.1.0>

Stacktrace:
[{gen_server,call,2,[{file,"../AtomVM/libs/estdlib/src/gen_server.erl"},{line,434}]}]

cp: #CP<module: 9, label: 18, offset: 16>

x[0]: error
x[1]: function_clause
x[2]: {1,1,60,1,[{4,1820}],error}

Stack 
-----

[]
[]
#CP<module: 1, label: 14, offset: 0>

Mailbox
-------

Monitors
--------
link to <0.2.0>
link to <0.8.0>
link to <0.14.0>

**End Of Crash Report**
Return value: error
I (3856) AtomVM: AtomVM application terminated.  Going to sleep forever ...
```

2. yes it's from atomvm (running on esp32-s3

3. i feel (../AtomVM/libs/estdlib/src/gen_server.erl"},{line,434}])
```
call(Name, Request, TimeoutMs) when is_atom(Name) ->
    case erlang:whereis(Name) of
        undefined ->
            erlang:exit({noproc, {?MODULE, ?FUNCTION_NAME, [Name, Request]}});
        Pid when is_pid(Pid) ->
            call(Pid, Request, TimeoutMs)
    end;
call(Pid, Request, TimeoutMs) when is_pid(Pid) ->
    MonitorRef = monitor(process, Pid),
    Pid ! {'$gen_call', {self(), MonitorRef}, Request},
    receive
        {'DOWN', MonitorRef, process, Pid, {E, []} = _Reason} ->
            erlang:exit({E, {?MODULE, ?FUNCTION_NAME, [Pid, Request]}});
        {'DOWN', MonitorRef, process, Pid, {_E, _L} = Reason} ->
            erlang:exit(Reason);
        {'DOWN', MonitorRef, process, Pid, Atom} when is_atom(Atom) ->
            erlang:exit({Atom, {?MODULE, ?FUNCTION_NAME, [Pid, Request]}});
        {MonitorRef, Reply} ->
            demonitor(MonitorRef, [flush]),
            Reply
    after TimeoutMs ->
        demonitor(MonitorRef, [flush]),
        erlang:exit({timeout, {?MODULE, ?FUNCTION_NAME, [Pid, Request]}})
    end.
```

cannot handle `{server, remote_node}`

elsbiet, 2025-12-09:
it seems to me that this issue has been resolved (had a look at the code but did not test (yet)). is this assumption correct