atomvm/AtomVM #1855: estdlib supervisor strategy one_for_all not implemented

karlsson, 2025-09-29:
According to the [documentation](https://doc.atomvm.org/main/apidocs/erlang/estdlib/supervisor.html#strategy) the `supervisor.erl` module implements `one_for_all` and `one_for_one` strategies but I can only find restart of the exiting child (one_for_one) implemented in the code and no check for alternative strategies.
The tests also seem to just use `one_for_one`.
Maybe it is just fine to implement one_for_one strategy for atomvm devices (for now), but then the documentation could be updated.

UncleGrumpy, 2025-10-02:
I believe this was an oversight by the author, thank you for raising the issue. We will try to add support for the `one_for_all` strategy in the near future.

UncleGrumpy, 2025-11-04:
Sorry it took so long to get a fix submitted, I found numerous other issues in the supervisor that I felt needed addressing, and I encountered some bugs in the test suite that I had to sort out in order to add tests for my fixes.

PR #1958 should take care of this issue, as well as a few others I opened as I discovered them (#1915 and #1957).

Thank you for opening the issue, this ended up leading a big improvement in our supervisor implementation. We should now be in alignment with OTP behavior for the features we do support.